tinkersprojects/G-Code-Arduino-Library #1: Read for Galvolaser

SerkanGuryalcin, 2019-03-22:
Hello everyone;

I am trying to do galvolaser and ı need to read g code. but ı have not any idea for how can I do it.
Any one can give any idea ? Thanks for everything.

tinkersprojects, 2019-03-23:
Hey mate, how/where is the g code
Is the gcode on an SD or is it getting sent to the controller

Regards,
William Bailes
https://tinkersprojects.com

SerkanGuryalcin, 2019-03-23:
I have Gcode from ınkspace raster 2 laser gcode and now I have text file include g code. I need to read g code and drive my motors but I don't know how can I do. Many g code examples for 3d printers and stepper motors but my motors are galvo. They can drive with voltege and scan angle.

tinkersprojects, 2019-03-23:
Ok, you will need to send it to the device through serial port
using universal g code sender or put it on a SD card. The Arduino
controller can almost handle the gcode, it will convert it to a double
value that you can use in your program. You will need to write the rest of
the program to control the motors. This library will not control the
motors.
I have been wanting to make a similar control unit But have not had time
too. If you been help with the program or understand in the Arduino library
let me know and I can see what I can do.

Regards,
William Bailes
https://tinkersprojects.com

SerkanGuryalcin, 2019-03-23:
Actually we couldn't do anything :(

tinkersprojects, 2019-03-23:
What's your hardware?

Regards,
William Bailes
https://tinkersprojects.com

SerkanGuryalcin, 2019-03-23:
I am using arduino nano. I can only scan square area with DAC and MCP4725 but I want do to process any image

tinkersprojects, 2019-03-25:
Did you want to send me your code and see what's going on.
Also if you want, you can contact me through discord or skype for a chat.

Regards,
William Bailes
https://tinkersprojects.com

closed 2019-05-17 by tinkersprojects