tinkersprojects/G-Code-Arduino-Library #2: Error message "no matching function for call to 'gcode::gcode(int, commandscallback [1])"

EmiShiraishi, 2019-05-17:
Hello 
I tried to use your awesome library to control the motors with G-code.
I had a problem in Arduino IDE when using your example files.
Below is the error message I got...

`gcode` Commands(1,commands);

> no matching function for call to 'gcode::gcode(int, commandscallback [1])'

[image]

I already installed <gcode.h> library. But somehow it doesn't work well.
Sorry for your time. If you have your time, please help me.

Thanks,
Emi Shiraishi

tinkersprojects, 2019-05-17:
hey, i will see whats going on and will get back to you soon

tinkersprojects, 2019-05-17:
hey, i have been working it and have not updated the git repository with the updates.
please redownload the library and try it again.

closed 2019-05-30 by tinkersprojects