tinkersprojects/G-Code-Arduino-Library #3: ERROR Examples X e Y

LeandroLimaPRO, 2019-10-30:
When you enter G28 X10 Y5 in position, in the coordinate examples 2 and 3, the data is not collected correctly, always gives the last string number sent in the variable X / newX. In the example above the return is: OK! X5.00 Y0.00

LeandroLimaPRO, 2019-10-31:
Resolvido

closed 2019-10-31 by LeandroLimaPRO

tinkersprojects, 2019-10-31:
Sorry I have not had time to review this.
Could you tell me about the issue more and how you had resolved, please.
if there is any way to make it better, just let me know.

LeandroLimaPRO, 2019-11-03:
Hello! I sent the correction by Git, it was the variable newX repeated twice, the second wrongly because the second collected the value that should go to Y

tinkersprojects, 2019-11-03:
hey
yea i saw that update you pushed through
could you contact me please, it would be interesting to know what you are
doing with it.
i have set an email to leo**to@hotmail.com

email me at tinkersprojects@gmail.com
or contact me with discord: https://discord.gg/knqCW4

LeandroLimaPRO, 2019-11-04:
I can get into the discord, but I understand very little English