tinkersprojects/G-Code-Arduino-Library #9: void homing and OnLED

ivaring, 2020-03-06:
Hi there,
while wanting to load sketch to Arduino Uno I get these error messages.

First error: " 'OnLED' was not declared on this scope"
and after declaring as a string
Second error: " expcected initializeer before 'void' " with cursor before void homing,

thanks.

tinkersprojects, 2020-03-06:
hey, 
are you using LED_Control. if so I have just fixed the syntax bug in the code, thank you for flagging it.

The fix is the functions need to be declared before commandscallback commands[NumberOfCommands] = {{"L1",OnLED},{"L2",OffLED}};
i have added 
void OnLED();
void OffLED(); 
before commands[NumberOfCommands]  line and removed void home()

ivaring, 2020-03-06:
Thank you very much for your fast feedback,
adding those lines the sketch worked,
one question I think I have to open serial monitor for sending L1 and L2 comands istn't?, if yes the LED is always off,

thanks again.

tinkersprojects, 2020-03-07:
can you also confirm that you are sending it as a new line.  down the bottom of the serial monitor in Arduino insure Newline is selected, its in the bottom right next to the bitrate.

closed 2020-09-30 by tinkersprojects