tinkersprojects/G-Code-Arduino-Library #11: Function moviment() is a member of a class

ProtopIazma, 2020-05-17:
Hi.
How do I write code correctly if the function moviment() is a member of a class? For example in this case:
**class Moving
{
  public:
    void gotoLocation(double f,double x,double y,double z,double e);
    void moviment();
} ArmMoving;
commandscallback commands[1] = {"G0",ArmMoving.moviment};
gcode Commands(1,commands);**

The compiler writes the following error:
**cannot convert 'Moving::moviment' from type 'void (Moving::)()' to type 'CallbackFunction {aka void (*)()}'**

tinkersprojects, 2020-05-18:
i'm looking into it but for the time being, you can just use a void function with the class function in it.
`
class Moving
{
public:
void gotoLocation(double f,double x,double y,double z,double e);
void moviment();
} ArmMoving;

void moviment()
{
ArmMoving.moviment
}

commandscallback commands[1] = {"G0",moviment};
gcode Commands(1,commands);
`
its a bit annoying but it will work until the other is worked out.

ProtopIazma, 2020-05-18:
Thank. I would be thankful. I tried to pass this function through the pointer, but then the compiler writes an error about type mismatch.

ProtopIazma, 2020-05-18:
Sorry, but the compiler for such a line:
**ArmMoving.moviment;**
Produces such an error: 
**invalid use of non-static member function 'void Moving::moviment()'**

closed 2020-09-30 by tinkersprojects