tinkersprojects/G-Code-Arduino-Library #14: How to send Gcode to arduino and use the callback system

nbaddorf, 2021-01-28:
Hi, I am trying to figure out how to send gcode to the arduino. I am using the universal gcode sender, but I am not sure what profile to use. I have tried all the profiles, but the grbl profile dosnt work, and the only profiles that seem to work is the tinyg and g2core profiles. When I use these profiles, the callback system dosnt seem to work. The callback messages show up in the console, but I don't have access to the jog settings or the current position of the tool. I am not sure what to do. Thanks!

tinkersprojects, 2021-01-29:
Hey sorry for the late reply
This library is just an interpreter, if you want current position and settings, you will need to write that your self. If you want this, you will want to use GRBL.
What are you trying to do with the library?
If you want, send me a message in discord. https://discord.gg/DhkwDNSP

nbaddorf, 2021-01-29:
I am trying to make a hanging wall pen plotter, so I wanted to use your library to read the gcode to position the system. I thought that I would need the Arduino to update the gcode sender when it reached its target, so that it would know to send the next command. I am not super familiar with the way gcode is sent, so forgive me if I am saying something totally wrong. Thanks for your help!

tinkersprojects, 2021-02-01:
yea you will be able to use this library for your application.
for what I know but also depends on the gcode sender, by sending "ok", "rs" or "!!" it saids that the Arduino as finished executing the function and to send the next. 
the updating part gets a bit bit more problematic and unknown. i think, i could be wrong but i think you send some like "!X=123" to update the position. i know tinyg and grbl have slitly different ways of doing it and it took me a while to workout this. 
most of the time i'm using this library from an SD card or a string but will need to research this more

nbaddorf, 2021-02-01:
Oh, that makes total sense! Thanks so much for your help!

closed 2021-02-01 by nbaddorf