tinkersprojects/G-Code-Arduino-Library #15: Only G Code with one number ist working

twinking92, 2021-02-12:
Hi,
i realized that only G-Codes with one number is working. For Example G-Codes like L1, L2, G1 are regonized, but L01 and L02 won't.

I used and modified your LED example.

```c
#include <gcode.h>

#define LEDpin 13
#define NumberOfCommands 2

void OnLED();
void OffLED();
// commandscallback commands[NumberOfCommands] = {{"L1",OnLED},{"L2",OffLED}}; // this workes
commandscallback commands[NumberOfCommands] = {{"L01",OnLED},{"L02",OffLED}}; // this does not
gcode Commands(NumberOfCommands,commands);

void setup()
{
  Commands.begin("ok"); //responce => ok, rs or !!
  pinMode(LEDpin, OUTPUT);
}

void loop() 
{
  Commands.available();
}

void OnLED()
{
  digitalWrite(LEDpin, HIGH);
}

void OffLED()
{
  digitalWrite(LEDpin, LOW);
}
```

tinkersprojects, 2021-02-18:
i'm looking into it
thank you

twinking92, 2021-02-19:
I did some further testing. I noticed that GCodes with two digits in general work. But if you use GCodes with a prefix "0" (like "G01"), this leads to problems.
The combination of "G01" and "G2" works. However, the combination of "G01" and "G02" does not. So the solution is not to define GCodes with preceding "0".
So instead of defining the GCodes "G01", "G02", etc. you have to define "G1", "G2". When entering GCodes, "G01" will still be recognized as "G1" which is exactly the desired behavior.

So the issue is solved I guess. Thanks for your great library.

closed 2022-02-03 by twinking92