tinkersprojects/G-Code-Arduino-Library #16: Galvo’s movement using the G-Code

jojoxyz, 2022-02-23:
Can anyone help me?
This is the code that controls Galvo's movement using the G-Code.
The base code works, but only in a simple form. Galvo jump between positions. It is not possible to control the speed.
I want Galvo to move smoothly. (Code part deactivated) It works in another code, but not in this one. Galvo doesn't move at all.
I don't know what I'm doing wrong.

`#include <gcode.h>
#include <MCP48xx.h>

MCP4822 dac(53);       // CS Pin 53

#define LDAC    48     // LDAC Pin
#define Laser   47     // Laser Pin

unsigned int x_i = 0;
unsigned int y_i = 0;
#define F_mSec = 10;

#define STEPS_MM 10   

unsigned float stepsx;
unsigned float stepsy;

double X;
double Y;

void moviment_0();
void moviment_1();
void gotoLocation();

#define NUMCOMMANDS 2

commandscallback commands[NUMCOMMANDS] = { { "G1",moviment_1} , {"G0", moviment_0} }; 
gcode Commands(NUMCOMMANDS,commands);

void setup()
{
    Commands.begin();

    pinMode      ( Laser ,  OUTPUT ) ;
    pinMode      ( LDAC  ,  OUTPUT ) ;
    digitalWrite ( LDAC  ,     LOW ) ;      
   
    dac.init();
    dac.turnOnChannelA();
    dac.turnOnChannelB();

    digitalWrite( Laser , HIGH );  
    delay(500);  
    digitalWrite( Laser , LOW );
}

void loop() { Commands.available(); }

void gotoLocation(double x,double y)
{ 
    stepsx = x*STEPS_MM;  // In value X mm to DAC Output  0 = 4000mV
    stepsy = y*STEPS_MM;  // In value Y mm to DAC Output  0 = 4000mV

/*
  //#### X
  if (stepsx  >= x_i ){x_i ++ ; if (x_i >= stepsx) x_i = stepsx; delayMicroseconds(F_mSec); } // Move X ++
 else  
  if (stepsx  <= x_i ){x_i -- ; if (x_i <= stepsx) x_i = stepsx; delayMicroseconds(F_mSec); } // Move X --

//#### Y
  if (stepsy  >= y_i ){y_i ++ ; if (y_i >= stepsy) y_i = stepsy; delayMicroseconds(F_mSec); } // Move Y ++
 else  
  if (stepsy  <= y_i ){y_i -- ; if (y_i <= stepsy) y_i = stepsy; delayMicroseconds(F_mSec); } // Move Y --

    dac.setVoltageA(x_i);       
    dac.setVoltageB(y_i);
    dac.updateDAC();
*/
    delay(10);
    
    dac.setVoltageA(stepsx);       
    dac.setVoltageB(stepsy);      
    dac.updateDAC();

}

//### G00
void moviment_0()
{ 
    double new_0_XValue = X;
    double new_0_YValue = Y;
    if(Commands.availableValue('X'))          // ADDED parameter X in G0
      new_0_XValue = Commands.GetValue('X');
    if(Commands.availableValue('Y'))          // ADDED parameter Y in G0
      new_0_YValue = Commands.GetValue('Y');

      digitalWrite( Laser , LOW );
      gotoLocation(new_0_XValue,new_0_YValue);
}

//### G01
void moviment_1()
{ 
    double new_1_XValue = X;
    double new_1_YValue = Y;
    if(Commands.availableValue('X'))          // ADDED parameter X in G1
      new_1_XValue = Commands.GetValue('X');
    if(Commands.availableValue('Y'))          // ADDED parameter Y in G1
      new_1_YValue = Commands.GetValue('Y');

      digitalWrite( Laser , HIGH );
      gotoLocation(new_1_XValue,new_1_YValue);
}`

tinkersprojects, 2022-02-24:
what is the Galvo’s device you are using?

i think it may be because your just setting your DAC and that will send it to a point, you may need to fade it.

jojoxyz, 2022-02-24:
20k  https://aws1.discourse-cdn.com/arduino/original/4X/0/e/3/0e303d1834c57b7b58928a0797db4e9d4a0d85de.jpeg

Now I've changed the code.
To stop reading a new line until the value of x_i and y_i is the same as stepsx and stepsy.
according to my gcode I draw a square.

G0 X10 Y10
G1 X200 Y10
G1 X200 Y200
G1 X10 Y200
G1 X10 Y10

But draw something for this. (Sorry, I have no way to insert the image and that's why I created it in g-code)

G0 X10 Y10
G1 X30 Y10
G1 X200 Y200
G1 X200 Y220
G1 X180 Y220
G1 X10 Y30
G1 X10 Y10

It looks like he started drawing the first line but doesn't finish drawing and then immediately jump to the second
`
void loop()
{ 
  mov();
  
  if ( read_new_line == true )  Commands.available(); 

  dac.setVoltageA(x_i);
  dac.setVoltageB(y_i);
  dac.updateDAC();  
  
}

void mov()
{
//#### X
   if ( stepsx  >= x_i )
 { x_i ++ ; delayMicroseconds(F_mSec); if (x_i >= stepsx) x_i = stepsx;  } // Move X ++
                       else  
                         { x_i -- ; delayMicroseconds(F_mSec); if (x_i <= stepsx) x_i = stepsx;  } // Move X --

//#### Y
  if ( stepsy  >= y_i ) 
 { y_i ++ ; delayMicroseconds(F_mSec); if (y_i >= stepsy) y_i = stepsy;  } // Move Y ++
                       else  
                         { y_i -- ; delayMicroseconds(F_mSec); if (y_i <= stepsy) y_i = stepsy;  } // Move Y --

  if ( ( x_i == stepsx ) && ( y_i == stepsy ))  read_new_line = true  ;  
  if ( ( x_i != stepsx ) && ( y_i != stepsy ))  read_new_line = false ; 

}

void gotoLocation(double x,double y)
{ 
    stepsx = x*STEPS_MM;  // In value X mm to DAC Output  0 = 4000mV
    stepsy = y*STEPS_MM;  // In value Y mm to DAC Output  0 = 4000mV

}
`

jojoxyz, 2022-02-24:
How can I pause reading from Buffer?

jojoxyz, 2022-02-24:
It works now.

`
void loop()
{ 
  mov();

  if ( ( x_i == stepsx ) && ( y_i == stepsy ))  Commands.available();      // !!!!!!!!!!!!!!!
 
  dac.setVoltageA(x_i);
  dac.setVoltageB(y_i);
  dac.updateDAC();  
  
}

void mov()
{
//#### X
   if ( stepsx  >= x_i ) { x_i ++ ; delayMicroseconds(F_mSec); if (x_i >= stepsx) x_i = stepsx;  } // Move X ++
                       else  
                         { x_i -- ; delayMicroseconds(F_mSec); if (x_i <= stepsx) x_i = stepsx;  } // Move X --

//#### Y
  if ( stepsy  >= y_i )  { y_i ++ ; delayMicroseconds(F_mSec); if (y_i >= stepsy) y_i = stepsy;  } // Move Y ++
                       else  
                         { y_i -- ; delayMicroseconds(F_mSec); if (y_i <= stepsy) y_i = stepsy;  } // Move Y --

}
`

tinkersprojects, 2022-02-24:
sorry been busy, but looks like you sorted it out. looks good
let us know what your making, i been wanting to play with one of these but have not had the chance

jojoxyz, 2022-02-24:
No problem.

I'm trying to create a firmware for laser engraving using Galvo.
The movement is almost resolved. I still have to make sure that the galvo will have a constant speed no matter how far apart the coordinates of the drawing are. Another problem is Unit Correction (mm). I can solve this, but the problem is that X0 is in the corner of the matrix. Somehow I have to turn the coordinates to make it work. Me and math, we're not good friends. Subsequently, the reduction of the laser power at a large change of direction must be solved.
Then it should work normally.
[image]

jojoxyz, 2022-02-24:
In the end, it would take a communication protocol so that my firmware can communicate with GRBL.
specifically with this version.
https://lasergrbl.com/

jojoxyz, 2022-02-24:
Unit correction (mm) is complete.
I won't publish it yet. People also need to work harder.
When I'm done, I can mail you my code. Because your library made my job a lot easier.

tinkersprojects, 2022-03-23:
thanks mate for the information

jojoxyz, 2022-03-24:
https://github.com/jojoxyz/Galvo_Laser_laser-engraver_Proj